nickmitchko/Octoprint-Filament-Reloaded #2: Print not pausing despite sensor working properly?

ssorgatem, 2020-09-29:
From an email by James Hydra
Hi;
Hoping you can help. I had this plugin already and it worked fine.
I upgraded to your new version (thanks for taking it over) and now it doesn't pause the print.
The dash icon works and changes color when the sensor doesn't have filament, and even turns yellow or orange when the filament runs out and the printer should pause, but the printer doesn't pause.
Everything is plugged in and set up the same as before.
I am using a sensor that also needs 3v for a signal light, but your ReadMe said it should only be pin and ground (The code is set to use the Raspberry Pi's internal Pull-Up resistors, so the switch should be between your detection pin and a ground pin), so I unplugged it from the 3v pin, but it still won't pause the printer when the sensor is triggered.
Any idea why it won't work?

Attachments you may need to help me:
Sensor plugged into GPIO24 (pin 18) and Ground (pin 20) as per this attached pin-out.
Also attached are screenshots of the settings in Octoprint.
"Settings 1" attachment are the exact settings that worked in the previous version except the "Board Pin Resistor Mode" which I left at your default.
Settings 2 attachment is the only gcode that would work previously. {I have watched YouTube tutorials that said to use different gcode in the "Job Paused" and "Job Resumed" sections (that I actually preferred to use), but only the ones in settings 2 would work.}

Thanks again for taking this abandoned plugin over, but please help me get it to work again.

ssorgatem, 2020-10-01:
https://github.com/ssorgatem/Octoprint-Filament-Reloaded/commit/a724cc540b04a7026599aa9c3b5d50ed4b61e49a fixes this

closed 2020-10-01 by ssorgatem

ManuelS383, 2020-10-01:
i've the same problem. I've installed the new version from today 2020-10-01 from the Plugin Manager and from URL (https://github.com/ssorgatem/Octoprint-Filament-Reloaded/archive/master.zip) but it's still not Pausing, after the runout Signal i waited 2 minutes bevor i canceled it manually.

octoprint.log: 
2020-10-01 16:04:15,980 - octoprint.plugins.filamentreload - INFO - The value of the pin is 1. No filame nt = False input = 24
2020-10-01 16:04:23,612 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to " Starting"
2020-10-01 16:04:23,621 - octoprint.printer.standard.job - INFO - Print job started - origin: local, pat h: OffsetEinstellen/OffsetEinstellenOhneG29.gcode, owner: _admin, user: _admin
2020-10-01 16:04:23,629 - octoprint.plugins.filamentreload - INFO - PrintStarted: Enabling filament sens or.
2020-10-01 16:04:23,636 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2020-10-01 16:04:23,650 - octoprint.util.comm - INFO - Changing monitoring state from "Starting" to "Pri nting"
2020-10-01 16:07:25,055 - octoprint.plugins.filamentreload - INFO - The value of the pin is 0. No filament = True input = 24
2020-10-01 16:07:25,309 - octoprint.plugins.filamentreload - INFO - The value of the pin is 0. No filament = True input = 24
2020-10-01 16:09:35,839 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Cancelling"
2020-10-01 16:09:35,846 - octoprint.printer.standard.job - INFO - Print job cancelled - origin: local, path: OffsetEinstellen/OffsetEinstellenOhneG29.gcode, owner: _admin, user: _admin
2020-10-01 16:09:35,848 - octoprint.plugins.filamentreload - INFO - PrintCancelled: Disabling filament sensor.
2020-10-01 16:09:35,870 - octoprint.plugins.filamentreload - INFO - PrintFailed: Disabling filament sensor.
2020-10-01 16:09:40,397 - octoprint.util.comm - INFO - Changing monitoring state from "Cancelling" to "Operational"

ManuelS383, 2020-10-01:
I reinstalled the plugin again (deinstall und install) now i get the following error wenn the status chanced:

2020-10-01 16:43:15,599 - octoprint.plugins.filamentreload - INFO - The value of the pin is 0. No filament = True input = 24
2020-10-01 16:43:15,600 - octoprint.cli.server - ERROR - Uncaught exception
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_filamentreload/__init__.py", line 243, in sensor_callback
    if not self.sensor_active():
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_filamentreload/__init__.py", line 177, in sensor_active
    return self.active
AttributeError: 'FilamentReloadedPlugin' object has no attribute 'active'

ManuelS383, 2020-10-01:
now its working. thx

Just before starting printing i get this error message in logfile:

2020-10-01 16:43:15,853 - octoprint.cli.server - ERROR - Uncaught exception
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_filamentreload/__init__.py", line 243, in sensor_callback
    if not self.sensor_active():
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_filamentreload/__init__.py", line 177, in sensor_active
    return self.active
AttributeError: 'FilamentReloadedPlugin' object has no attribute 'active'