nickmitchko/Octoprint-Filament-Reloaded #3: Filament runout detection but not pause send

Disttrack, 2020-10-02:
I have this plugin and the detection of runout its correctly but the plugin never send the pause to the printer.

octoprint.log:

2020-10-02 12:39:00,977 - octoprint.plugins.filamentreload - INFO - The value of the pin is 0. No filament = True input = 4
2020-10-02 12:39:01,227 - octoprint.plugins.filamentreload - INFO - The value of the pin is 0. No filament = True input = 4
2020-10-02 12:39:01,477 - octoprint.plugins.filamentreload - INFO - The value of the pin is 0. No filament = True input = 4
2020-10-02 12:39:01,728 - octoprint.plugins.filamentreload - INFO - The value of the pin is 0. No filament = True input = 4
2020-10-02 12:39:01,978 - octoprint.plugins.filamentreload - INFO - The value of the pin is 0. No filament = True input = 4
2020-10-02 12:39:02,228 - octoprint.plugins.filamentreload - INFO - The value of the pin is 0. No filament = True input = 4
2020-10-02 12:39:02,478 - octoprint.plugins.filamentreload - INFO - The value of the pin is 0. No filament = True input = 4
2020-10-02 12:39:02,728 - octoprint.plugins.filamentreload - INFO - The value of the pin is 0. No filament = True input = 4
2020-10-02 12:39:02,978 - octoprint.plugins.filamentreload - INFO - The value of the pin is 0. No filament = True input = 4
2020-10-02 12:39:03,228 - octoprint.plugins.filamentreload - INFO - The value of the pin is 0. No filament = True input = 4

ssorgatem, 2020-10-03:
Did you update tot he last version?
I fixed this issue in the last release.
Unless there's yet another issue triggering the same (mis)behaviour...

Could you attach a "clean" octoprint.log?

By clean I mean:
Erase the current one, then restart octoprint.
Then, start a dummy print and trigger the filament sensor.

You can find the log file in the "Logging" section of Octoprint's settings.

Disttrack, 2020-10-04:
> Did you update tot he last version?
> I fixed this issue in the last release.
> Unless there's yet another issue triggering the same (mis)behaviour...
>
> Could you attach a "clean" octoprint.log?
>
> By [...]

Yes, I updated to latest version, but detects "no filament" state but not pause the print. When I can I upload a octoprint.log.

Thanks!

ManuelS383, 2020-10-04:
Hi, i've got the same Problem but not with the first print after reboot. I had tested the sensor after rebooting the Pi an it works fine, then i've started my print and it doesn't stop. After that i tested it several times, everytime the first printjob works great (also multiple breaks after each other) but after canceling the job. The next Job send the correct signal but did not pause the print.

smcallis77, 2020-10-05:
> Hi, i've got the same Problem but not with the first print after reboot. I had tested the sensor after rebooting the Pi an it works fine, then i've started my print and it doesn't stop. After that i tested it several times, everytime the first printjob works great (also multiple breaks after each other) but after canceling the job. The next Job send the correct signal but did not pause the print.
> 
> I've made this BugFix for me local:
> octoprint_filamentreload/**init**.py
> Row 236: self.active = 0 -> #self.active = 0
> It seems that the reactivation by starting a new print did not work correcly so i never deactivate it.

It seems likely I have the same problem. Tested the plugin on a single print and the printer paused correctly. However, in production after printing multiple jobs the filament ran out and the printer did not pause.

ssorgatem, 2020-10-06:
Mmm, this is strange, I've been trying to reproduce this but unsuccessfully.

ManuelS383, 2020-10-06:
I am using: 
OctoPrint 1.4.2 Python 3.7.3 OctoPi 0.17.0

Hope this helps. If you need any more Information please tell me.

ssorgatem, 2020-10-06:
I just pushed a new release.
It works for me... could you test if it fixes this issue?

ssorgatem, 2020-10-06:
> I am using:
> OctoPrint 1.4.2 Python 3.7.3 OctoPi 0.17.0
> 
> Hope this helps. If you need any more Information please tell me.

Exactly the same here.
Maybe it has to do with the sensor and how it behaves? 
I'm using a repurposed z-stop switch (I installed a BLTouch)...

smcallis77, 2020-10-07:
> > I am using:
> > OctoPrint 1.4.2 Python 3.7.3 OctoPi 0.17.0
> > Hope this helps. If you need any more Information please tell me.
> 
> Exactly the same here.
> Maybe it has to do with the sensor and how it behaves?
> I'm using a repurposed z-stop switch (I installed a BLTouch)...

I haven't tried the new release, will try later today. In the meantime though:
OctoPrint 1.4.2 Python <not sure> 2.7.x I think OctoPi 0.16.0

ssorgatem, 2020-10-08:
any news? Can i close this issue?

smcallis77, 2020-10-09:
Sorry for late reply. my printer went belly up and its taken a while to get it working again. The fix in 1.3.3 looks OK so far. I have been able to trigger a pause multiple times within a single print and also trigger multiple pauses across print jobs. To me it looks like it is fixed. I will continue to monitor.

commit adae9807 (2023-01-30): Fix #3 pullup
closed 2020-10-09 by ssorgatem