nickmitchko/Octoprint-Filament-Reloaded #4: No filament recognized although there is one

PuddingBaer91, 2020-10-05:
My print won't start because "no filament is detected" but there is filament loaded. Please see my log
[octoprint (3).log](https://github.com/ssorgatem/Octoprint-Filament-Reloaded/files/5328037/octoprint.3.log)

ssorgatem, 2020-10-06:
The new version should fix this. Let me know if this error persists.

PuddingBaer91, 2020-10-06:
this error is with version 1.3.2 (already the newest one)

ssorgatem, 2020-10-13:
The newest one is 1.3.3.

PuddingBaer91, 2020-10-17:
sorry i meant 1.3.3

ssorgatem, 2020-10-17:
can you provide a log file with the newest version?

PuddingBaer91, 2020-10-18:
The log i provided in the first post is from version 1.3.3

ssorgatem, 2020-10-19:
No, it isn't, unless you are a time traveller ;)
Last line int he log is from 2020-10-**05** 15:32:21
Version 1.3.3 is from 2020-10-**06** 15:18:00

It came out 23 hours, 14 minutes and 21 seconds after your log file.
Which even accounting for a +/-13 hours timezone difference,  means your log is from an older version ;)

Try with the newest one.

dexer12, 2020-10-22:
i got a simmilar issue. i installed the current 1.3.3 version of you plugin.
I setup everything as i think it should be fine. I selected the GPIO Pin with the Switch which is ussualy closed, when filament is running thru.
I used BCM Mode so it is using GPIO Pinout and i selected pull up resistor. for testing i also added an M117 GCode so it would send a Message if the switch will get triggered. but its not working.
I'm using a small microswitch, like the one of the endstops connected on pin to GPIO 24 and the other to Ground.
If i try it with the "Filament Sensor Simplified" it is working like expected.
But i really would like to use your plugin, because it has some nice features, like the icon in the navbar or the cancelation of prints, i try to start without filament available. thats really really great!

ssorgatem, 2020-10-23:
Have you tried changing the pullup/pulldown resistor setting?
"Filament Sensor Simplified"  has no way to change that and it wouldn't work for my sensor, which worked for the deault configuration of this plugin....

And that's why I ended up adopting it :P

Also, it'd be really helpful if you could provide a log file.

dexer12, 2020-10-23:
Yeah i changed the pullup/pulldown resistor settings.
Do you know which one "Filament Sensor Simplified" uses?
If i check the status with the /plugin/filamentreload/status page it is always "0".
What logfile information do you need?

ssorgatem, 2020-10-23:
The octoprint.log file.
Ideally "clean", that is, just after rebooting octoprint, and maybe try triggering the sensor.

nickmitchko, 2023-03-06:
Closing as stale, if this is still an issue, please file another issue or reopen :)

closed 2023-03-06 by nickmitchko