nickmitchko/Octoprint-Filament-Reloaded #5: Filament runout sensor icon not updating in touchui.

smcallis77, 2020-10-09:
My filament sensor is working well. Using the chrome interface (non touchui) the runout sensor updates in the navbar as per below.
[image]
 However, in the touchui interface the icon shows as per below.

[image]

ssorgatem, 2020-10-09:
I suspect this may be due to touchui changing the icons?
I really have no idea.
I opened an issue on TouchUI's github to get some help

nickmitchko, 2022-01-29:
Touch UI applies their own class overrides to these jinja templates. It should be as easy as adding in the class overrides into the jinja template. I'll take a look

nickmitchko, 2022-01-29:
Update; we can define a less variable to work with OctoprintUI without much change to the plugin or JS.

Example here: [PSU Control Less Variables](https://github.com/kantlivelong/OctoPrint-PSUControl/blob/master/octoprint_psucontrol/static/less/psucontrol.less)

nickmitchko, 2023-03-06:
I believe the latest version has these changes for touchUI to work. Let me know if it doesn't.

closed 2023-03-06 by nickmitchko