nickmitchko/Octoprint-Filament-Reloaded #6: Question: Supported Filament Sensors

joshuajmlee, 2020-10-10:
I can't find any lists or any documentation anywhere to answer this question, so I'm hoping this is a good place to ask. Does anyone have a list of sensors that the raspberry pi and octoprint support for this plugin?

ssorgatem, 2020-10-13:
Anything that opens or closes a circuit is supported.
You just need to wire it properly and configure it accordingly (the pullup/pulldown resistor setting + whether it is a normally open or normally closed sensor).

joshuajmlee, 2020-10-13:
> Anything that opens or closes a circuit is supported.
> You just need to wire it properly and configure it accordingly (the pullup/pulldown resistor setting + whether it is a normally open or normally closed sensor).

Perfect, thank you!

closed 2020-10-13 by joshuajmlee