nickmitchko/Octoprint-Filament-Reloaded #7: 1.3.3 Uncaught Exception when setting parameters: object has no attribute 'active'

dan-and, 2020-10-29:
Hi, 

I tried out 1.3.3 with Octoprint 
`2020-10-29 08:58:03,179 - octoprint.environment - INFO - Detected environment is Python 3.7.3 under Linux (linux). Details:
|  hardware:
|      cores: 4
|      freq: 1500.0
|      ram: 1903980544
|  os:
|      id: linux
|      platform: linux
|  plugins:
|      pi_support:
|          model: Raspberry Pi 4 Model B Rev 1.1
|          octopi_version: 0.17.0
|  python:
|      pip: 20.2.3
|      version: 3.7.3
|      virtualenv: /home/pi/oprint
`

`2020-10-29 09:43:33,037 - octoprint.plugins.filamentreload - INFO - Setting up sensor.
2020-10-29 09:43:33,038 - octoprint.plugins.filamentreload - INFO - Using BCM Mode
2020-10-29 09:43:33,039 - octoprint.plugins.filamentreload - INFO - Filament Sensor active on GPIO Pin [24]
2020-10-29 09:43:33,040 - octoprint.plugins.filamentreload - INFO - Filament Sensor Pin uses pulldown
2020-10-29 09:43:33,040 - octoprint.plugins.filamentreload - INFO - Setting new checkrate
2020-10-29 09:43:53,954 - octoprint.plugins.filamentreload - INFO - The value of the pin is 0. No filament = True input = 24
2020-10-29 09:43:53,964 - octoprint.cli.server - ERROR - Uncaught exception
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_filamentreload/__init__.py", line 236, in sensor_callback
    if not self.active:
AttributeError: 'FilamentReloadedPlugin' object has no attribute 'active'
2020-10-29 09:45:08,908 - octoprint.plugins.filamentreload - INFO - Setting up sensor.
2020-10-29 09:45:08,909 - octoprint.plugins.filamentreload - INFO - Using BCM Mode
2020-10-29 09:45:08,910 - octoprint.plugins.filamentreload - INFO - Filament Sensor active on GPIO Pin [24]
2020-10-29 09:45:08,911 - octoprint.plugins.filamentreload - INFO - Filament Sensor Pin uses pullup
2020-10-29 09:45:08,911 - octoprint.plugins.filamentreload - INFO - Setting new checkrate
`

Thanks

ssorgatem, 2020-10-30:
But, does it work?
Are you by chance checking the filament status through the API just after restarting Octoprint?

ssorgatem, 2020-11-02:
Can you try installing this version of the plugin and see if it still happens?
Just paste the link into Octoprint's plugin manager where you can paste a link to download a plugin:

https://github.com/ssorgatem/Octoprint-Filament-Reloaded/archive/c1793aa197e15534c983628a7b3c416232c5e24a.zip

woogerboy21, 2020-11-17:
> Can you try installing this version of the plugin and see if it still happens?
> Just paste the link into Octoprint's plugin manager where you can paste a link to download a plugin:
> 
> https://github.com/ssorgatem/Octoprint-Filament-Reloaded/archive/c1793aa197e15534c983628a7b3c416232c5e24a.zip

I ran into this issue and loaded this via the method you mention.  It seems to resolve the issue reported and python crashing but now states there is an upgrade available to the plugin so you may want to make sure any version information gets updated before publishing things (if that is how the version checking works).

ssorgatem, 2020-11-18:
Yes, the version issue was intented, that's just a testing branch.

I'll push the fix then.

closed 2020-11-18 by ssorgatem