nickmitchko/Octoprint-Filament-Reloaded #12: Issue an m600?

jtannenba, 2021-04-29:
When my printer runs out of filament, it just notifies Octoprint and stops, making it almost impossible to remove the filament.  Can we issue a m600 or other commands to unload the filament?

nickmitchko, 2022-01-29:
@jtannenba Out of Filament GCODE can issue any GCODE to the printer when you run out of filament. I'd advise against issuing an [m600](https://marlinfw.org/docs/gcode/M600.html) in-case you are not near the printer at that moment.

[image]

jtannenba, 2022-01-29:
This was a long time ago.  Everything is working fine now.
Thank you

closed 2022-01-29 by jtannenba