nickmitchko/Octoprint-Filament-Reloaded #13: Pull Up Resistor causes OP Crash

charredchar, 2021-05-30:
OctoPrint 1.6.1
OctoPi 0.18.0

Any time I select Pull Up Resistor in the settings OctoPrint crashes and needs to be restarted via SSH.

Log doesn't show anything... but it is attached. Enclosure Plugin does not cause this crash when setting up the same pin the same way.

https://pastebin.com/ySMEQt2g

`pi@octopi:~ $ sudo service octoprint status
● octoprint.service - The snappy web interface for your 3D printer
   Loaded: loaded (/etc/systemd/system/octoprint.service; enabled; vendor preset: enabled)
   Active: failed (Result: signal) since Sun 2021-05-30 10:13:28 CDT; 4min 59s ago
  Process: 357 ExecStart=/home/pi/oprint/bin/octoprint serve --host=${HOST} --port=${PORT} (code=killed, signal=SEGV)
 Main PID: 357 (code=killed, signal=SEGV)

May 30 10:13:27 octopi octoprint[357]: 2021-05-30 10:13:27,134 - octoprint.plugins.enclosure - INFO - Setting GPIO pin 12 as OUTPUT with initial value: 1
May 30 10:13:27 octopi octoprint[357]: 2021-05-30 10:13:27,134 - octoprint.plugins.enclosure - INFO - Setting GPIO pin 18 as OUTPUT with initial value: 1
May 30 10:13:27 octopi octoprint[357]: 2021-05-30 10:13:27,149 - octoprint.plugins.filamentreload - INFO - Setting up sensor.
May 30 10:13:27 octopi octoprint[357]: 2021-05-30 10:13:27,150 - octoprint.plugins.filamentreload - INFO - Using Board Mode
May 30 10:13:27 octopi octoprint[357]: 2021-05-30 10:13:27,151 - octoprint.plugins.filamentreload - INFO - Filament Sensor active on GPIO Pin [15]
May 30 10:13:27 octopi octoprint[357]: 2021-05-30 10:13:27,152 - octoprint.plugins.filamentreload - INFO - Filament Sensor Pin uses pullup
May 30 10:13:27 octopi octoprint[357]: 2021-05-30 10:13:27,153 - octoprint.plugins.filamentreload - INFO - Setting new checkrate
May 30 10:13:27 octopi octoprint[357]: 2021-05-30 10:13:27,199 - octoprint.plugins.filamentreload - DEBUG - Thread: Update icon 1
May 30 10:13:28 octopi systemd[1]: octoprint.service: Main process exited, code=killed, status=11/SEGV
May 30 10:13:28 octopi systemd[1]: octoprint.service: Failed with result 'signal'.
`

robbrad, 2021-08-21:
These are my settings 
[image]

Oddly I dont see **Setting new checkrate** in my log

nickmitchko, 2022-01-29:
@robbrad is this issue still open? I can take a look. 

### 1. Try this first:
- Wire the sensor pin to Pin 24 
[image]
- Change the "Board Pin Mode" Setting to "Board Mode"

### 2. If that doesn't work:
List the following here:
- the sensor you're using
- picture of your pi and sensor

robbrad, 2023-01-30:
Prob a question for @charredchar - as my instance is working fine

enz1ey, 2023-02-14:
> OctoPrint 1.6.1 OctoPi 0.18.0
>
> Any time I select Pull Up Resistor in the settings OctoPrint crashes and needs to be restarted via SSH.
>
> Log doesn't show anything... but it is attached. Enclosure [...]

I was having this issue with the stock filament sensor on the Ender 6. As soon as I'd connect the sensor with no filament loaded, the Pi would crash because it was shorting VCC and GND pins.

I solved it by connecting VCC to a ground pin and the yellow signal wire to GPIO 26, then setting the plugin to normally open.

closed 2023-03-06 by nickmitchko