nickmitchko/Octoprint-Filament-Reloaded #15: Does not compile

Mesmer88, 2023-01-23:
Trying to install this on latest octopi + octoprint (1.8.6) but getting this error and plugin gets disabled automatically:

```
2023-01-23 19:59:42,627 - octoprint.plugin.core - WARNING - Plugin Octoprint-filamentreload (1.4.0) can't be compiled under Python 3.7.3 due to invalid syntax
2023-01-23 19:59:42,663 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2023-01-23 19:59:42,707 - octoprint.plugin.core - ERROR - Invalid syntax in plugin file of plugin filamentreload
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugin/core.py", line 782, in _parse_metadata
    return parse_plugin_metadata(self.location)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugin/core.py", line 78, in parse_plugin_metadata
    root = ast.parse(f.read(), filename=path)
  File "/usr/lib/python3.7/ast.py", line 35, in parse
    return compile(source, filename, mode, PyCF_ONLY_AST)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_filamentreload/__init__.py", line 183
    return {
           ^
TabError: inconsistent use of tabs and spaces in indentation
```

robbrad, 2023-01-30:
@nickmitchko - I get a new error now with this version ( happy to open a new issue)

`2023-01-30 22:45:13,510 - octoprint.cli.server - ERROR - Uncaught exception
Traceback (most recent call last):
  File "/home/pi/oprint/bin/octoprint", line 8, in <module>
    sys.exit(main())
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/__init__.py", line 936, in main
    octo(args=args, prog_name="octoprint", auto_envvar_prefix="OCTOPRINT")
  File "/home/pi/oprint/lib/python3.7/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/home/pi/oprint/lib/python3.7/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/pi/oprint/lib/python3.7/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/pi/oprint/lib/python3.7/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/click/decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/cli/server.py", line 330, in serve_command
    disable_color=no_color,
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/cli/server.py", line 163, in run_server
    octoprint_server.run()
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/server/__init__.py", line 653, in run
    self._setup_blueprints()
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/server/__init__.py", line 1815, in _setup_blueprints
    ) = self._prepare_blueprint_plugins()
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/server/__init__.py", line 1862, in _prepare_blueprint_plugins
    blueprint, prefix = self._prepare_blueprint_plugin(plugin)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/server/__init__.py", line 1913, in _prepare_blueprint_plugin
    blueprint = plugin.get_blueprint()
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1688, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugin/types.py", line 1486, in get_blueprint
    f = getattr(self, member)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_filamentreload/__init__.py", line 93, in pullup
    return int(self._settings.get(["pullup"] or 0))
TypeError: int() argument must be a string, a bytes-like object or a number, not 'NoneType'
`

robbrad, 2023-01-30:
Should this 
https://github.com/nickmitchko/Octoprint-Filament-Reloaded/blob/9ec4020513625450f6097274dd61ade759a7a909/octoprint_filamentreload/__init__.py#L93

be :

` return int(self._settings.get(["pullup"] ,0))`

nickmitchko, 2023-01-30:
Thanks for spot checking this, I'll reopen and test on my instance

robbrad, 2023-01-30:
My understanding is incorrect its not an or value on the mixin I think it just needs a default adding im testing now

robbrad, 2023-01-30:
@nickmitchko : This makes the plugin work (well OP loads at least) https://github.com/robbrad/Octoprint-Filament-Reloaded/commit/3f1ed06b6601f01cf2f2b86d7df967e62bec5c61

But is doing this the desired behaviour?

nickmitchko, 2023-01-30:
@robbrad Just need to determine if that makes pullup always 0

robbrad, 2023-01-30:
OK Confirmed it works on my system - it can be chamged in settings (but it initialised as 0 for first install)

robbrad, 2023-01-30:
[image]

Shows out of Filament ! which is good because it is !

robbrad, 2023-01-30:
and with filament 
[image]

robbrad, 2023-01-31:
@nickmitchko - I just tested 1.4.1 and unfortunatly the same issue happens on clean install `  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_filamentreload/__init__.py", line 93, in pullup
    return int(self._settings.get(["pullup"])) or 0
TypeError: int() argument must be a string, a bytes-like object or a number, not 'NoneType'
`

nickmitchko, 2023-01-31:
Odd behavior as there must be a change in octoprint between 1.8.6 and 1.8.1. I took your PR since all the code changes work on my test printer (octopi 1.8.1 and python 3.7.3).

Testing again now as 1.4.2.

robbrad, 2023-01-31:
1.4.2 works for me

nickmitchko, 2023-01-31:
Nice, glad this is fixed up.

closed 2023-01-31 by nickmitchko