nickmitchko/Octoprint-Filament-Reloaded #18: Uninitialized variable.

sillyfunnypedro, 2023-03-10:
@nickmitchko, thanks for taking this on and continuing to support.

I was playing around with this a while back and noticed that the variable self.active is not initialized i had some spurious failures (i cannot remember now, it was a while back). I think its wise to initialize self.active in the self.initialize() method.  I see no harm in initializing the variable, but if i am wrong please educate me.  
Thanks
@sillyfunnypedro

Here is the code that i am running

```    def initialize(self):
        self._logger.info(
            "Running RPi.GPIO version '{0}'".format(GPIO.VERSION))
        if GPIO.VERSION < "0.6":       # Need at least 0.6 for edge detection
            raise Exception("RPi.GPIO must be greater than 0.6")
        GPIO.setwarnings(False)        # Disable GPIO warnings
        self.pin_value = -1			   # Cache the pin value when we detect out of filament
        self.active = 0. #<---- this initialization helped my spurious failures.  
```

nickmitchko, 2023-03-13:
Thanks, I'll take a look and see if the code you suggests breaks anything in my test instance. Might be a few weeks until I get around to it.

sillyfunnypedro, 2023-05-23:
I also did the same testing and could not get a failure.  I tested this a few versions ago and the error is no longer duplicateable.

closed 2023-05-23 by sillyfunnypedro