nickmitchko/Octoprint-Filament-Reloaded #23: Filament issue machine error octoprint 1.8.7

omgitsmatt, 2024-10-30:
hi i had a problem with this, octoprint 1.8.7 python 3.9.2 octoPi 1.0.0
I installed it and set the sensor to 1 because it was on -1 and just when the filament was out of the sensor the machine return to the start point and give me error. Can you help me please? I also give you a log file
@nickmitchko
[octoprint filament log.log](https://github.com/user-attachments/files/17571939/octoprint.filament.log.log)

nickmitchko, 2024-10-31:
Hello @omgitsmatt,

Can you take a screenshot of your settings in the octoprint UI?

There's a few possibilities here:
1. Somehow the setting for the pin-mode got dropped
2. The pin you are using is not gpio compatible or the pin is used by another plugin
3. The limit switch is wired incorrectly

omgitsmatt, 2024-10-31:
hi @nickmitchko thank you for the tempestive answer, i send you the scrrenshot of octoprint settings ui 
[image]

omgitsmatt, 2024-10-31:
@nickmitchko the pin was on -1 before the print and i set it to 1

omgitsmatt, 2024-11-07:
@nickmitchko

nickmitchko, 2024-11-07:
Can you take a screenshot of the settings pane, something simple is off.

omgitsmatt, 2024-12-12:
@nickmitchko what settings pane? what section? i send you this screen that are the settings of the plugin
[image]

nickmitchko, 2025-04-08:
@omgitsmatt make sure when you check the settings that you test the pin (should show in the top of the status bar) with the filament available and out by putting a small piece of filament through your sensor

omgitsmatt, 2025-04-09:
[image] Hi Nick i changed the pin to 18 and now the filament icon (i guess this is the filament icon) is orange but if i put the filament in the sensor it doesn't change, is always orange. I tried also to put the pin another time to 1 only to see if was orange because of the right pin but it's always orange now, also with the pin 1, before it ever wasn't orange

omgitsmatt, 2025-04-09:
@nickmitchko how can i know the right pin?

nickmitchko, 2025-04-09:
The right pin is the pin where you plug your sensor in your GPIO pin array on the RPI